Mieze/AtherosE2200Ethernet #4: Sporadic total failure of kext

mattnoir, 2016-09-20:
Hi,
this may not be perfectly described issue as I have no idea what causes it. It happens second time today and issue is that after turning PC off, when I turned it back on on second day, ethernet was not recognized at all and nothing helped to solve this issue. Previous time it somehow worked when I installed ALXethernet back and then deleted and replaced with this kext, but not this time. If you need some specific settings or something, ask what you need.

I am on El Capitan, Lenovo G580 20160

Mieze, 2016-09-20:
I need kernel logs showing the incident in order to find out what is going on.

mattnoir, 2016-09-20:
Okay, how do I get it? Sorry, never did it before

Mieze, 2016-09-20:
Please see: http://www.insanelymac.com/forum/topic/300056-solution-for-qualcomm-atheros-ar816x-ar817x-and-killer-e220x/

mattnoir, 2016-09-20:
Here
[kernel log.zip](https://github.com/Mieze/AtherosE2200Ethernet/files/483499/kernel.log.zip)

Mieze, 2016-09-21:
The driver fails to get the NIC's MAC address as the error message states:

Ethernet [AtherosE2200]: Failed to get permanent MAC address.

AFAIK this is either a hardware issue or due to a wrong BIOS setting but I don't know how to resolve it.

mattnoir, 2016-09-21:
The funny thing is, it shows this error even on ALXEthernet. So 1. I screwed up when changing kext or 2. there is something weird going on.
Because I changed literally no settings, nothing in BIOS and one day it worked, second day it stopped working

mattnoir, 2016-09-27:
UPDATE:
After some time with ALXEthernet I switched back and it works again. I changed nothing and there is no error regarding ethernet in system log. ¯\(°_o)/¯

closed 2018-01-24 by Mieze