Mieze/AtherosE2200Ethernet #5: Qualcomm Atheros Killer E2200 is down after random time of system uptime

AlexTKS, 2016-09-23:
Hello.
Sorry about my english. English is not my native language, but I try to explain.
I install macOS Sierra and I had a problem after that

My MB is msi z87-g45 gaming. 
Network is

Qualcomm Atheros Killer E2200 PCI Express Gigabit Ethernet:

  Имя:	ethernet
  Тип:	Ethernet
  Шина:	PCI
  Слот:	Ethernet
  ID производителя:	0x1969
  ID устройства:	0xe091
  ID производителя подсистемы:	0x1462
  ID подсистемы:	0x7821
  Идентификатор версии:	0x0013
  Ширина соединения:	x1
  Имя BSD:	en0
  Имя Kext:	AtherosE2200Ethernet.kext
  Размещение:	/System/Library/Extensions/AtherosE2200Ethernet.kext
  Версия:	1.1.1

Network interface works well some time after each boot. But every boot after random time interface is down. I have to reboot system to fix.

logs in down time

------------------------------
Ethernet [AtherosE2200]: Timer fired while link down.
Ethernet [AtherosE2200]: EEE link partner: 0x0006.
Ethernet [AtherosE2200]: Link up on en0, 1-Gigabit, full-duplex, Rx/Tx flow-control, energy-efficient-ethernet
Ethernet [AtherosE2200]: EEE link partner: 0x0006.
Ethernet [AtherosE2200]: Fatal interrupt. Reseting chip. ISR=0x2000200
Ethernet [AtherosE2200]: Timer fired while link down.
Ethernet [AtherosE2200]: EEE link partner: 0x0006.
Ethernet [AtherosE2200]: Link up on en0, 1-Gigabit, full-duplex, Rx/Tx flow-control, energy-efficient-ethernet
Ethernet [AtherosE2200]: EEE link partner: 0x0006.
Ethernet [AtherosE2200]: Fatal interrupt. Reseting chip. ISR=0x2000200
Ethernet [AtherosE2200]: Timer fired while link down.
Ethernet [AtherosE2200]: EEE link partner: 0x0006.
Ethernet [AtherosE2200]: Link up on en0, 1-Gigabit, full-duplex, Rx/Tx flow-control, energy-efficient-ethernet
Ethernet [AtherosE2200]: EEE link partner: 0x0006.
Ethernet [AtherosE2200]: Fatal interrupt. Reseting chip. ISR=0x2000200
Ethernet [AtherosE2200]: Timer fired while link down.
Ethernet [AtherosE2200]: EEE link partner: 0x0006.
Ethernet [AtherosE2200]: Link up on en0, 1-Gigabit, full-duplex, Rx/Tx flow-control, energy-efficient-ethernet
Ethernet [AtherosE2200]: EEE link partner: 0x0006.
Ethernet [AtherosE2200]: Fatal interrupt. Reseting chip. ISR=0x2000200
Ethernet [AtherosE2200]: Timer fired while link down.
Ethernet [AtherosE2200]: EEE link partner: 0x0006.
Ethernet [AtherosE2200]: Link up on en0, 1-Gigabit, full-duplex, Rx/Tx flow-control, energy-efficient-ethernet
Ethernet [AtherosE2200]: EEE link partner: 0x0006.
Ethernet [AtherosE2200]: Fatal interrupt. Reseting chip. ISR=0x2000200
Ethernet [AtherosE2200]: Timer fired while link down.
Ethernet [AtherosE2200]: EEE link partner: 0x0006.
Ethernet [AtherosE2200]: Link up on en0, 1-Gigabit, full-duplex, Rx/Tx flow-control, energy-efficient-ethernet
Ethernet [AtherosE2200]: EEE link partner: 0x0006.
Ethernet [AtherosE2200]: Fatal interrupt. Reseting chip. ISR=0x2000200
Ethernet [AtherosE2200]: Timer fired while link down.
Ethernet [AtherosE2200]: EEE link partner: 0x0006.
Ethernet [AtherosE2200]: Link up on en0, 1-Gigabit, full-duplex, Rx/Tx flow-control, energy-efficient-ethernet
Ethernet [AtherosE2200]: EEE link partner: 0x0006.
Ethernet [AtherosE2200]: Fatal interrupt. Reseting chip. ISR=0x2000200
Ethernet [AtherosE2200]: Timer fired while link down.
Ethernet [AtherosE2200]: EEE link partner: 0x0006.
Ethernet [AtherosE2200]: Link up on en0, 1-Gigabit, full-duplex, Rx/Tx flow-control, energy-efficient-ethernet
Ethernet [AtherosE2200]: EEE link partner: 0x0006.
Ethernet [AtherosE2200]: Fatal interrupt. Reseting chip. ISR=0x2000200
Ethernet [AtherosE2200]: Timer fired while link down.
[... 1053 lines cut ...]
setMulticastList() ===>
setMulticastList() <===
setMulticastList() ===>
setMulticastList() <===
setMulticastList() ===>
setMulticastList() <===
setMulticastList() ===>
setMulticastList() <===
setMulticastList() ===>
setMulticastList() <===
Ethernet [AtherosE2200]: Fatal interrupt. Reseting chip. ISR=0x2000200
txClearDescriptors() ===>
txClearDescriptors() <===
setMulticastMode() ===>
setMulticastMode() <===
getPacketFilters() ===>
getPacketFilters() <===
setMulticastList() ===>
setMulticastList() <===
setMulticastList() ===>
setMulticastList() <===
setMulticastList() ===>
setMulticastList() <===

I hope you solve the problem.

Mieze, 2016-09-23:
This is a hardware bug. Version 2.2.0 implements a workaround for it. By the way, usage of version 1.x.x is not recommended anymore.

AlexTKS, 2016-09-23:
But this hardware woks well on OS X EL Capitan and OS Windows. I do not understand this. Adapter worked 3 years with no errors. On Windows works well now too.

AlexTKS, 2016-09-23:
I download fast commit and compile. I sure that I install version 2.2.0

Mieze, 2016-09-23:
The problem is a hardware issue with certain receive buffer address patterns. For a description of the problem please see: https://git.kernel.org/cgit/linux/kernel/git/next/linux-next.git/commit/drivers/net/ethernet/atheros/alx?id=881d0327db37ad917a367c77aff1afa1ee41e0a9

closed 2016-09-23 by Mieze