Mieze/AtherosE2200Ethernet #6: Quick disconnect / reconnect periodically

tomatlscomm, 2016-12-02:
Hello Mieze.

Thank you for this great kext.
However i have an issue with it.

Sometimes (without doing anything special) the connection is lost, ans immediatly back.
I have this log about this problem : 

```
Dec  2 15:51:30 Toms-iMac kernel[0]: Ethernet [AtherosE2200]: Link down on en2
Dec  2 15:51:30 Toms-iMac kernel[0]: in6_unlink_ifa: IPv6 address 0xa95adc8b8428148f has no prefix
Dec  2 15:51:33 toms-imac kernel[0]: Ethernet [AtherosE2200]: Link up on en2, 1-Gigabit, full-duplex, Rx/Tx flow-control, energy-efficient-ethernet
```

Any idea ?

Regards

Vladislav-Hloboshin, 2016-12-02:
Hello.
I had this problem and fixed it by disabling 'energy-efficient-ethernet' feature.
Try this https://www.dropbox.com/s/0kpn96vioeffv73/Screenshot%202016-12-02%2018.00.35.png?dl=0 (see Duplex field)

tomatlscomm, 2016-12-02:
Hello.

Thank you for your answer.
OK i just tried that and let you know if the problem appear again or if your solution solves it

Regards

closed 2017-01-30 by Mieze