Mieze/AtherosE2200Ethernet #8: How to "Recreate the kernel cache"

rutcreate, 2017-01-20:
I tried to follow the installation step but I don't know how 2) Recreate the kernel cache and 5) Install the new driver and recreate the kernel cache. Could you please explain more about them?

Thanks in advance.

Mieze, 2017-01-20:
Use Kext Wizard (or a similar kext tool) to install the driver as it performs the required actions on its own. Choose "Installation", select the kext, click "Install" and wait for the action to finish. Done!

rutcreate, 2017-01-20:
I have KextBeast now but how do I get .kext?

P.S. I'm really new here.

Mieze, 2017-01-20:
Please see: http://www.insanelymac.com/forum/files/file/313-atherose2200ethernet/

rutcreate, 2017-01-20:
Thanks for prompt reply. Finally I got kext file. Big thanks.

By the way, may I know what is the xcode project in this repository for? I believe that it is for build .kext file but I don't know how to. Do you have any resource for doing this?

Mieze, 2017-01-20:
It's the source code of the kext so that you may build it yourself. Want to build it yourself? Google knows the answer! ;-)

rutcreate, 2017-01-20:
Haha. I know google knows that too. Thanks for reply anyway. I think this should be closed now.

closed 2017-01-20 by rutcreate