Mieze/AtherosE2200Ethernet #9: Disconnects when transferring large files to NAS

pengi77, 2017-01-29:
Hello, I am experiencing an issue where when I try to transfer a large file from my computer to the NAS (900MB - 1GB), the LAN connection in Sierra goes down. Has anyone else had an issue similar to this?

coolspot18, 2017-05-21:
Yes, I've had a similar issue, it used to happen more often with Yosemite and El Capitan... but since moving to Clover it happens much rarer.

However, I did have a hard reset just now; I was running a speed test (900mbit/s) on speedtest.net, I disabled my Ethernet driver (Make Inactive in Network Panel) and OSX panicked and reset itself.

Mieze, 2017-05-21:
I can't reproduce the issue. Usually kernel panics are the result of a messed up system (system caches) or a broken DSDT (floating regions which haven't been patched properly by Clover, other DSDT errors).

closed 2018-01-24 by Mieze