Mieze/AtherosE2200Ethernet #10: Link is always down

polarathene, 2017-02-20:
I've got this NIC but I'm on Linux with the `alx` kernel driver. I've been exhaustively for the past 3 days tried to debug what is wrong with the NIC. I've got a cat5e that works with other machines to establish a connection to a router, this doesn't work with the laptops `Killer E220x [1969:e091] (rev 13)`.

I'm curious if you could provide any insight that might be causing this issue with your experience developing this macOS driver. Could it be that Windows driver previously may have done something that I cannot reverse on the Linux end? I'm running Arch with the 4.9.9 kernel. The laptop has an intel wireless device that works perfectly, the issue is specific to this ethernet controller.

I'd appreciate any advice :)

Mieze, 2017-02-20:
On some machines the BIOS leaves the NIC in power state D3 so that the alx driver fails to initialize the chip. I was faced with this problem during development too but managed to resolve the issue with a call to 

IOReturn AtherosE2200::setPowerStateWakeAction(OSObject *owner, void *arg1, void *arg2, void *arg3, void *arg4)
{
    AtherosE2200 *ethCtlr = OSDynamicCast(AtherosE2200, owner);
    IOPCIDevice *dev;
    UInt16 val16;
    UInt8 offset;
    
    if (ethCtlr) {
        dev = ethCtlr->pciDevice;
        offset = ethCtlr->pciPMCtrlOffset;
        
        val16 = dev->extendedConfigRead16(offset);
        
        val16 &= ~(kPCIPMCSPowerStateMask | kPCIPMCSPMEStatus | kPCIPMCSPMEEnable);
        val16 |= kPCIPMCSPowerStateD0;
        
        dev->extendedConfigWrite16(offset, val16);
    }
    return kIOReturnSuccess;
}

which restores the chips's power state to D0 in order to make it fully operational.

polarathene, 2017-02-23:
I appreciate the help, althoough I'm not sure if I'm able to put it to good use myself. I've tried to reach out to others who might be able to assist :) I'm guessing that would need to be a patch to the alx kernel driver?

You wouldn't happen to know if I can use an app or terminal command to query what the current power state is or also change it would you? Many users of this NIC on MSI forums have issues, it seems it can possibly be fixed if I reinstall Windows and change some settings via the driver there. No idea if that'll carry over to fixing it on Linux however.

Mieze, 2017-02-23:
In case you aren't able to implement a patch yourself, you could try to contact the Linux driver maintainers (see alx driver source on www.kernel.org), explain the problem to them and point them to this conversation. They might be willing to help you and fix the driver. Good luck!

closed 2017-02-23 by Mieze