Mieze/AtherosE2200Ethernet #12: Wake On Lan not working

Lawris, 2017-11-22:
Kext Version : 2.2.2
Device : Qualcomm Atheros Killer E2200
Os : macOS Sierra (10.12.1)
Motherboard : MSI H97 Gaming 3

Hello, I can't get WOL working on my hackintosh. 
I think the BIOS config is good, I allowed wake up by PCI-E, wake on lan works when I shutdown from the windows partition but if I shutdown from macOS, it's like the wol packet is ignored.

I wanted to troubleshoot but I don't know where to activate the "debug mode" (maybe clover but did not find there ?).

Any help would be greatly appreciated, thanks.

Mieze, 2017-11-22:
It's not a bug, it's the specified behavior. The driver doesn't support WoL from S5 because there is no support for this feature in macOS.

Lawris, 2017-11-22:
Thanks for replying, excuse my noobiness but what is "S5" ?

Mieze, 2017-11-22:
S5 = soft off

By the way, don't get me wrong. I'm not trying to stop anybody from doing anything. The driver is open source and you are free to modify it according to your needs, provided you have the ability to do it. I just wanted to explain why I designed it this way.

closed 2017-11-22 by Lawris

jockebq, 2019-06-06:
I use this kext in Mojave and I am having the same issue. Using the Z97MX-Gaming 5 motherboard with E2201 ethernet.
You say it is designed to do this? Is it because Mac OS have no setting in the OS for wake on lan, like on Windows?
How would you modify the kext to enable Wake On Lan?