Mieze/AtherosE2200Ethernet #14: Sometimes sleep causes a restart

Vladislav-Hloboshin, 2018-02-16:
Anonymous UUID:       52E2348D-7A23-8D27-4AA4-93F7F8696D05

Fri Feb 16 11:59:06 2018

*** Panic Report ***
panic(cpu 4 caller 0xffffff80189ff085): Kernel trap at 0xffffff7f99462993, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0xffffff93c4d9c1dc, CR3: 0x000000002632e000, CR4: 0x00000000003627e0
RAX: 0xffffff805d4f2400, RBX: 0xffffff8357c02000, RCX: 0xffffff93c4d9c000, RDX: 0x0000000000000100
RSP: 0xffffff93c647bcd0, RBP: 0xffffff93c647bd70, RSI: 0x00000000000001d0, RDI: 0xffffff8357c02000
R8:  0x0000000000000000, R9:  0xffffff801900b058, R10: 0x0000000000000000, R11: 0xffffff83b85ad058
R12: 0xffffff7f9943299c, R13: 0xffffff8357c02000, R14: 0xffffff8357c02000, R15: 0x0000000000000000
RFL: 0x0000000000010202, RIP: 0xffffff7f99462993, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0xffffff93c4d9c1dc, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0, VF: 1

Backtrace (CPU 4), Frame : Return Address
0xffffff93c647b960 : 0xffffff80188e950c 
0xffffff93c647b9e0 : 0xffffff80189ff085 
0xffffff93c647bbc0 : 0xffffff801889a543 
0xffffff93c647bbe0 : 0xffffff7f99462993 
0xffffff93c647bd70 : 0xffffff7f99462956 
0xffffff93c647bd90 : 0xffffff7f994329d4 
0xffffff93c647bdc0 : 0xffffff8018ebddda 
0xffffff93c647be30 : 0xffffff7f99432a3b 
0xffffff93c647be80 : 0xffffff7f99436c7a 
0xffffff93c647bef0 : 0xffffff8018b843c5 
0xffffff93c647bfb0 : 0xffffff80188998f7 
      Kernel Extensions in backtrace:
         com.apple.iokit.IONetworkingFamily(3.2)[6326DB88-5330-3F0C-91F6-D478AB5E7503]@0xffffff7f9942a000->0xffffff7f99456fff
         com.insanelymac.AtherosE2200Ethernet(2.2.1)[64DD0A8F-6C61-3332-B6E8-59BD7B8C1CA6]@0xffffff7f9945e000->0xffffff7f99465fff
            dependency: com.apple.iokit.IONetworkingFamily(3.2)[6326DB88-5330-3F0C-91F6-D478AB5E7503]@0xffffff7f9942a000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[116F5E39-E2DA-3211-86A2-5F01E8B07E01]@0xffffff7f9904e000

BSD process name corresponding to current thread: kernel_task
Boot args: dart=0 darkwake=8 nvda_drv=1 kext-dev-mode=1 uia_exclude=HS09;HS10;SS04;SS05;SS06;SS09;SS10 -igfxoff 

Mac OS version:
16G1212

Kernel version:
Darwin Kernel Version 16.7.0: Thu Jan 11 22:59:40 PST 2018; root:xnu-3789.73.8~1/RELEASE_X86_64
Kernel UUID: 181BE5F8-44DA-3654-8C62-A41A005038D7
Kernel slide:     0x0000000018600000
Kernel text base: 0xffffff8018800000
__HIB  text base: 0xffffff8018700000
System model name: iMac17,1 (Mac-B809C3757DA9BB8D)

System uptime in nanoseconds: 2036494687742
last loaded kext at 1999332738650: as.lvs1974.IntelGraphicsFixup	1.2.3 (addr 0xffffff7f9c0e4000, size 16384)
last unloaded kext at 873356247249: as.lvs1974.IntelGraphicsFixup	1.2.3 (addr 0xffffff7f9c0e4000, size 16384)
loaded kexts:
as.lvs1974.IntelGraphicsFixup	1.2.3
org.virtualbox.kext.VBoxNetAdp	5.2.6
org.virtualbox.kext.VBoxNetFlt	5.2.6
org.virtualbox.kext.VBoxUSB	5.2.6
org.virtualbox.kext.VBoxDrv	5.2.6
com.intel.kext.intelhaxm	6.0.1
net.sf.tuntaposx.tap	1.0
net.sf.tuntaposx.tun	1.0
com.nvidia.web.NVDAGM100HalWeb	10.1.8
com.nvidia.web.GeForceWeb	10.1.8
com.nvidia.web.NVDAResmanWeb	10.1.8
com.intel.driver.EnergyDriver	2.0
com.insanelymac.AtherosE2200Ethernet	2.2.1
as.vit9696.Shiki	2.2.0
as.vit9696.Lilu	1.2.1
org.netkas.driver.FakeSMC	1759
com.rehabman.driver.USBInjectAll	0.6.2
com.apple.filesystems.afpfs	11.0.5
com.apple.nke.asp-tcp	8.0.1
com.apple.filesystems.autofs	3.0
com.apple.driver.AudioAUUC	1.70
com.apple.driver.AGPM	110.23.17
com.apple.driver.ApplePlatformEnabler	2.7.0d0
com.apple.driver.X86PlatformShim	1.0.0
com.apple.driver.AppleHDA	279.48
com.apple.driver.pmtelemetry	1
com.apple.iokit.IOUserEthernet	1.0.1
com.apple.driver.AppleUpstreamUserClient	3.6.4
com.apple.driver.AppleMCCSControl	1.3.4
com.apple.iokit.IOBluetoothSerialManager	5.0.5f1
com.apple.Dont_Steal_Mac_OS_X	7.0.0
com.apple.driver.AppleHV	1
com.apple.driver.AppleIntelPCHPMC	1.1
com.apple.driver.AppleIntelSlowAdaptiveClocking	4.0.0
com.apple.driver.AppleOSXWatchdog	1
com.apple.iokit.BroadcomBluetooth20703USBTransport	5.0.5f1
com.apple.AppleFSCompression.AppleFSCompressionTypeDataless	1.0.0d1
com.apple.AppleFSCompression.AppleFSCompressionTypeZlib	1.0.0
com.apple.BootCache	40
com.apple.filesystems.hfs.kext	366.70.3
com.apple.iokit.IOAHCIBlockStorage	295.20.1
com.apple.driver.AirPort.Brcm4360	1152.2.1a2
com.apple.driver.AppleAHCIPort	326.60.1
com.apple.driver.AppleACPIButtons	5.0
com.apple.driver.AppleRTC	2.0
com.apple.driver.AppleHPET	1.8
com.apple.driver.AppleSMBIOS	2.1
com.apple.driver.AppleAPIC	1.7
com.apple.nke.applicationfirewall	172
com.apple.security.quarantine	3
com.apple.security.TMSafetyNet	8
com.apple.security.SecureRemotePassword	1.0
com.apple.iokit.IOUSBUserClient	900.4.1
com.apple.kext.triggers	1.0
com.apple.driver.DspFuncLib	279.48
com.apple.kext.OSvKernDSPLib	525
com.apple.iokit.IOSurface	159.12
com.apple.driver.AppleSMBusController	1.0.18d1
com.apple.iokit.IOSMBusFamily	1.1
com.apple.iokit.IOAcceleratorFamily2	311.16.2
com.apple.AppleGraphicsDeviceControl	3.14.49
com.apple.driver.AppleSSE	1.0
com.apple.iokit.IONDRVSupport	516.1
com.apple.driver.AppleHDAController	279.48
com.apple.iokit.IOGraphicsFamily	515.3
com.apple.iokit.IOHDAFamily	279.48
com.apple.iokit.IOSlowAdaptiveClockingFamily	1.0.0
com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport	5.0.5f1
com.apple.iokit.IOBluetoothHostControllerUSBTransport	5.0.5f1
com.apple.iokit.IOBluetoothHostControllerTransport	5.0.5f1
com.apple.iokit.IOBluetoothFamily	5.0.5f1
com.apple.driver.X86PlatformPlugin	1.0.0
com.apple.driver.IOPlatformPluginFamily	6.0.0d8
com.apple.iokit.IOSerialFamily	11
com.apple.driver.AppleSMC	3.1.9
com.apple.iokit.IOSCSIArchitectureModelFamily	394.50.1
com.apple.driver.AppleUSBAudio	308.7
com.apple.iokit.IOAudioFamily	205.15
com.apple.vecLib.kext	1.2.0
com.apple.driver.AppleUSBHostMergeProperties	1.1
com.apple.driver.usb.IOUSBHostHIDDevice	1.1
com.apple.driver.usb.networking	5.0.0
com.apple.driver.usb.AppleUSBHostCompositeDevice	1.1
com.apple.filesystems.hfs.encodings.kext	1
com.apple.driver.usb.AppleUSBHostPacketFilter	1.0
com.apple.iokit.IO80211Family	1200.12.2
com.apple.driver.mDNSOffloadUserClient	1.0.1b8
com.apple.driver.corecapture	1.0.4
com.apple.iokit.IONetworkingFamily	3.2
com.apple.iokit.IOUSBFamily	900.4.1
com.apple.iokit.IOAHCIFamily	288
com.apple.driver.usb.AppleUSBXHCIPCI	1.1
com.apple.driver.usb.AppleUSBXHCI	1.1
com.apple.driver.AppleEFINVRAM	2.1
com.apple.driver.AppleEFIRuntime	2.1
com.apple.iokit.IOHIDFamily	2.0.0
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleKeyStore	2
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.driver.AppleCredentialManager	1.0
com.apple.driver.KernelRelayHost	1
com.apple.iokit.IOUSBHostFamily	1.1
com.apple.driver.AppleBusPowerController	1.0
com.apple.driver.DiskImages	444.50.16
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOReportFamily	31
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.driver.AppleACPIPlatform	5.0
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.Libm	1
com.apple.kec.pthread	1
com.apple.kec.corecrypto	1.0
Model: iMac17,1, BootROM IM171.0148.B00, 4 processors, Intel Core i7, 4,01 GHz, 32 GB, SMC 2.33f10
Graphics: NVIDIA GeForce GTX 980, NVIDIA GeForce GTX 980, PCIe, 4095 MB
Memory Module: BANK 0/DIMM0, 16 GB, DDR4, 2400 MHz, Crucial Technology, BLS16G4D240FSB.16FAD
Memory Module: BANK 1/DIMM0, 16 GB, DDR4, 2400 MHz, Crucial Technology, BLS16G4D240FSB.16FAD
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x133), Broadcom BCM43xx 1.0 (7.21.171.133.1a2)
Bluetooth: Version 5.0.5f1, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en2
PCI Card: NVIDIA GeForce GTX 980, Display Controller, PCI Slot 0
Serial ATA Device: Samsung SSD 850 PRO 256GB, 256,06 GB
Serial ATA Device: Samsung SSD 750 EVO 500GB, 500,11 GB
USB Device: USB 3.0 Bus
USB Device: USB Advanced Audio Device
USB Device: Bluetooth USB Host Controller
USB Device: USB Device
USB Device: USB Device
USB Device: USB 3.1 Bus
Thunderbolt Bus:

Mieze, 2018-02-16:
It's either a messes up system or a DSDT/BIOS related problem but I can rule out a driver error.

Vladislav-Hloboshin, 2018-02-16:
OK. Thank you.

closed 2018-06-06 by Mieze