Mieze/AtherosE2200Ethernet #15: Not matched with device

lightingghost, 2018-04-20:
I found the kext is not matched with my E2200, here is the output from `ioreg`, on `IOService`:

```
AtherosE2200  <class AtherosE2200, id 0x100000242, !registered, !matched, active, busy 0, retain 5>
    | |   |         {
    | |   |           "IOClass" = "AtherosE2200"
    | |   |           "CFBundleIdentifier" = "com.insanelymac.AtherosE2200Ethernet"
    | |   |           "IOProviderClass" = "IOPCIDevice"
    | |   |           "IOLinkStatus" = 0
    | |   |           "IOPowerManagement" = {"CapabilityFlags"=0,"CurrentPowerState"=0}
    | |   |           "IOLinkSpeed" = 0
    | |   |           "IOProbeScore" = 1000
    | |   |           "Driver_Version" = "2.2.2"
    | |   |           "IOPCIMatch" = "0x10901969 0x10911969 0x10A01969 0x10A11969 0xE0911969 0xE0A11969 0xE0B11969"
    | |   |           "maxIntrRate" = 6500
    | |   |           "IOMatchCategory" = "IODefaultMatchCategory"
    | |   |           "IOSelectedMedium" = ""
    | |   |           "enableTSO4" = Yes
    | |   |           "enableTSO6" = Yes
    | |   |           "enableCSO6" = Yes
    | |   |           "rxPolling" = Yes
    | |   |           "IOActiveMedium" = ""
    | |   |         }
```

On `IODevice`:
```
    | | +-o PXSX@0  <class IOPCIDevice, id 0x1000001ce, registered, matched, active, busy$
    | |     {
    | |       "assigned-addresses" = <1000038200000000000010ef000000000000040018000381000$
    | |       "IOInterruptSpecifiers" = (<1200000007000000>,<0500000000000100>)
    | |       "class-code" = <00000200>
    | |       "IODeviceMemory" = (({"address"=4010803200,"length"=262144}),"IOSubMemoryDe$
    | |       "built-in" = <00>
    | |       "subsystem-vendor-id" = <49180000>
    | |       "IOPowerManagement" = {"CapabilityFlags"=258,"CurrentPowerState"=2,"ChildPr$
    | |       "pci-aspm-default" = 0
    | |       "acpi-device" = "IOACPIPlatformDevice is not serializable"
    | |       "IOInterruptControllers" = ("io-apic-0","IOPCIMessagedInterruptController")
    | |       "name" = <"ethernet">
    | |       "IOChildIndex" = 1
    | |       "device-id" = <91e00000>
    | |       "vendor-id" = <69190000>
    | |       "acpi-pmcap-offset" = 64
    | |       "IOPCIResourced" = Yes
    | |       "compatible" = <"pci1849,e091","pci1969,e091","pciclass,020000","PXSX">
    | |       "IOPCIExpressLinkCapabilities" = 523281
    | |       "acpi-path" = "IOACPIPlane:/_SB/PCI0@0/RP03@1c0002/PXSX@0"
    | |       "subsystem-id" = <91e00000>
    | |       "revision-id" = <10000000>
    | |       "IOPCIExpressLinkStatus" = 4113
    | |       "IOPCIExpressCapabilities" = 1
    | |       "pcidebug" = "3:0:0"
    | |       "IOName" = "ethernet"
    | |       "acpi-wake-type" = 2
    | |       "reg" = <000003000000000000000000000000000000000010000302000000000000000000$
    | |     }
```

I saw in the [IOKit Fundamentals](https://developer.apple.com/library/content/documentation/DeviceDrivers/Conceptual/IOKitFundamentals/Matching/Matching.html#//apple_ref/doc/uid/TP0000015-TP9), driver need to implement `probe` to match with the device. But I cannot find any probe functions in this repo. 

I am new to this. I would appreciate it if you could help.

Mieze, 2018-04-20:
What's your problem? There is nothing wrong with the IORegistry entries.

lightingghost, 2018-04-20:
Thanks for helping. The problem is the ethernet card is not showing in my system. The IORegistry of the driver shows `not registered`, `not matched`; while the ethernet device shows `registered` and `matched`.

Mieze, 2018-04-20:
You might want to create the ethernet device in the Network pane of system preferences.

lightingghost, 2018-04-20:
I don't have that computer right at my hand. But if I remember correctly, there is no such choice in system preference. Besides, in the system information, under the ethernet cards, it shows nothing; and there is no ethernet service under the network.

Mieze, 2018-04-20:
Of course it's not there because You need to create it.

lightingghost, 2018-04-20:
I will have a try after getting home. Thanks then.

lightingghost, 2018-04-21:
[image]
Here is the create service panel. Note that "USB 10/100" is my USB ethernet card, not the built-in one. I don't think it is possible to create a service without the device being properly drivered.

closed 2019-12-29 by Mieze