Mieze/AtherosE2200Ethernet #17: Losing network access after sleep.

andreluisos, 2018-07-26:
Happened suddenly (probably after updating to 2.2.2).

Mieze, 2018-07-26:
Fix your system because it's not a driver issue. I'm using version 2.2.2 on my primary machine for work and never lost network connectivity.

andreluisos, 2018-07-30:
What would you suggest?

Thanks!

andreluisos, 2018-07-31:
2.2.1 also seems to be working good. I'll keep testing.

ganthore, 2018-07-31:
I'm running on the absolute latest version of this driver (ref: https://github.com/Mieze/AtherosE2200Ethernet/commit/a23d958057fc2dd121892149dff790a86d4a2879) and I am NOT having any problems with it after coming out of a sleep state.

andreluisos, 2018-07-31:
I'm back to the latest version to try again. Maybe it was a bad build or something like that. I'll let you know.

andreluisos, 2018-11-16:
Found out trying to build the kext myself with xcodebuild gives a bad kext, which causes the sleep issue.

What are the steps to build it?

Mieze, 2018-11-17:
Use Xcode 8 or 9 to build the kext. From the "Products" menu choose "Archive" and export the product (save into a directory). You might also want to try the prebuilt binary of version 2.2.2 which I posted a few moments ago.

andreluisos, 2018-11-18:
Thanks! I'll be closing this.

closed 2018-11-18 by andreluisos