Mieze/AtherosE2200Ethernet #19: Wake On Lan not working from either sleep nor shutdown

jockebq, 2019-06-11:
I have found this thread: https://github.com/Mieze/AtherosE2200Ethernet/issues/12
Which informs me that in S5 state (completely powered off) Wake On Lan should not work (as is the design).
But for me Wake On Lan doesn't work on either a completely powered off computer or sleep.

So I went into the code looking for answers, and found this:
`wolCapable = false;` inside AtherosE2200Ethernet.cpp

And with some googling I found this issue on another kext:
https://github.com/Mieze/IntelMausiEthernet/issues/17#issuecomment-424290662
Which says changing the values
```
wolCapable = true;
wolActive = true;
```
to true should solve this.
I found no wolActive, but I found wolCapable, and I changed it to true.

But it made no difference at all, it is not waking up from sleep or shutdown.
I have made sure that ErP is disabled in BIOS, and Wake On Lan is working fine from Windows 10.

hiradyazdan, 2019-11-21:
@jockebq have you made sure "Wake for Network access" box is checked in Energy Saver settings?

Mieze, 2019-12-29:
WoL is working fine on my machine. Im case you are having trouble, it might also be related to the DSDT, etc.

closed 2019-12-29 by Mieze

kaienli, 2020-11-12:
I changed wolCapable = true ， it is not waking up from shutdown.

Mieze, 2020-11-12:
Once again, WoL from S5 isn't supported by the driver because macOS doesn't support this feature. As the driver is published under the GPL, feel free to modify the code according to your needs.