Mieze/AtherosE2200Ethernet #20: Wake on Lan from sleep works, but next sleep is broken (instant wakeup)

jockebq, 2019-06-16:
Hi,

I have Wake On LAN from sleep working great. Except one issue.
After I have woken the hackintosh with wake on lan, and put it to sleep again, it will immidiatly wake up. This never happens when I have not used Wake On LAN to wake it up. So it is related to this driver. Somehow the Wake On LAN package is "saved" even after the computer has been woken up.

Waking up with Wake On LAN gives me this in the logs:
`Wake reason: PEG1 PEGP PEG2 PEGP PXSX PXSX (Network)`
That works good, but when putting it to sleep and it wakes up immidiatly, the wake reason looks like this:
`Wake reason: ?`

Is there any solution for this?

Thank you!

Mieze, 2019-06-16:
If the wakeup event had been generated by the NIC, you see the following line in your log file:

`Wake reason: PEG1 PEGP PEG2 PEGP PXSX PXSX (Network)`

but you have the line

`Wake reason: ?`

which means that there hasn't been any wakeup event by the the NIC but something else is preventing the system to sleep. It's not the driver!

jockebq, 2019-06-16:
Yes! I have that line in the log. As I stated in the first post, I wake up my computer via WOL and the log shows:
Wake reason: PEG1 PEGP PEG2 PEGP PXSX PXSX (Network)

Then I put it to sleep again, and this time it will immediately wake up, and now the log shows Wake reason: ?.

And this only happens after using wake on lan. So sleep is working great, wake up via keyboard works, wake on lan works. But after using wake on lan it breaks the next sleep.

Mieze, 2019-06-16:
...but it isn't caused by the driver. I think another service is prohibiting the machine from falling asleep again.

jockebq, 2019-06-16:
@Mieze
But it has no issue at all going to sleep at any time, except after using wake on lan. So there must be a connection. I asked around in the Hackintosh discord and some guys told me that there was a possibility that the wake on lan "command" were not cleared after being woken up.

Isn't there anything I can try? I have ruled out all other possibilities and I am out of ideas. Please help me solve this, it is the only thing missing from my Hackintosh that I really need working.

Mieze, 2019-06-16:
It is cleared on wakeup and even if it wasn't, the kernel log would show the NIC as wakeup reason anyway but it doesn't. You are definitely on the wrong track!

jockebq, 2019-06-16:
But why would it only happen after using wake up lan? How could it be something else? As that would mean sleep would never work fine?

Mieze, 2019-06-16:
Use 'pmset -g log' in Terminal to debug sleep issues.

jockebq, 2019-06-16:
Here is the log:

`Sleep/Wakes since boot at 2019-06-16 16:42:21 +0200 :4   Dark Wake Count in this sleep cycle:1

Time stamp                Domain              	Message                                                                    	Duration  	Delay     
==========                ======              	=======                                                                    	========  	=====     
UUID: 000C92B2-BCC0-4BA2-A32E-CEFD202919EA
2019-06-16 20:22:41 +0200 Notification        	Display is turned off                                                      	          
2019-06-16 20:22:41 +0200 Assertions          	PID 1075(rcd) Summary UserIsActive "com.apple.rcdevent" 01:58:45  id:0x0x900008d00 [System: DeclUser BGTask kDisp]          
2019-06-16 20:22:41 +0200 Assertions          	PID 1075(rcd) Summary UserIsActive "com.apple.rcdevent" 01:58:36  id:0x0x900008d0d [System: DeclUser BGTask kDisp]          
2019-06-16 20:22:41 +0200 Assertions          	PID 93(hidd) Summary UserIsActive "com.apple.iohideventsystem.queue.tickle.4294968377.3" 00:00:00  id:0x0x9000080f0 [System: DeclUser BGTask kDisp]          
2019-06-16 20:22:41 +0200 Assertions          	PID 312(cloudphotosd) Summary BackgroundTask "com.apple.cloudphotosd.darkwake.sync" 00:04:49  id:0x0xb00008de0 [System: DeclUser BGTask kDisp]          
2019-06-16 20:22:41 +0200 Assertions          	Kernel Idle sleep preventers: -None-          
2019-06-16 20:22:41 +0200 Assertions          	PID 53(powerd) Created InternalPreventSleep "com.apple.powermanagement.darkwakelinger" 00:00:00  id:0x0xd00008e8a [System: PrevIdle DeclUser BGTask SRPrevSleep kCPU kDisp]          
2019-06-16 20:22:41 +0200 Assertions          	PID 312(cloudphotosd) Released BackgroundTask "com.apple.cloudphotosd.darkwake.sync" 00:04:49  id:0x0xb00008de0 [System: PrevIdle DeclUser BGTask SRPrevSleep kCPU kDisp]          
2019-06-16 20:22:56 +0200 Assertions          	PID 53(powerd) TimedOut InternalPreventSleep "com.apple.powermanagement.darkwakelinger" 00:00:15  id:0x0xd00008e8a [System: DeclUser BGTask SRPrevSleep kCPU kDisp]          
2019-06-16 20:22:56 +0200 Sleep               	Entering Sleep state due to 'Software Sleep pid=87': Using AC (Charge:0%) 21 secs   
2019-06-16 20:22:59 +0200 PM Client Acks      	Delays to Sleep notifications: [com.apple.apsd is slow(2882 ms)]           
2019-06-16 20:23:11 +0200 Kernel Client Acks  	Delays to Sleep notifications: [AppleHDADriver driver is slow(msg: SetState to 0)(1012 ms)] [AMDFramebufferCI driver is slow(msg: SetState to 1)(2094 ms)] [AMDFramebufferCI driver is slow(msg: SetState to 1)(2142 ms)] [en0 driver is slow(msg: WillChangeState to 0)(2356 ms)]           
2019-06-16 20:23:11 +0200 Assertions          	Kernel Idle sleep preventers: IODisplayWrangler          
2019-06-16 20:23:11 +0200 Notification        	Display is turned on                                                       	          
2019-06-16 20:23:11 +0200 Assertions          	PID 202(mDNSResponder) Created MaintenanceWake "mDNSResponder:maintenance" 00:00:00  id:0x0xd00008e9f [System: PrevIdle DeclUser BGTask SRPrevSleep kCPU kDisp]          
2019-06-16 20:23:11 +0200 Assertions          	PID 202(mDNSResponder) Released MaintenanceWake "mDNSResponder:maintenance" 00:00:00  id:0x0xd00008e9f [System: PrevIdle DeclUser BGTask kDisp]          
2019-06-16 20:23:17 +0200 Wake                	Wake from Normal Sleep [CDNVA] due to PEG1 PEGP PEG2 PEGP PXSX PXSX/UserActivity Assert: Using AC (Charge:0%) 4 secs    
2019-06-16 20:23:17 +0200 HibernateStats      	hibmode=0 standbydelay=86400                                               	          109       	
2019-06-16 20:23:17 +0200 WakeTime            	WakeTime: 7.007 sec                                                        	          
2019-06-16 20:23:17 +0200 Kernel Client Acks  	Delays to Wake notifications: [AppleHDADriver driver is slow(msg: SetState to 1)(1453 ms)] [PRT1 driver is slow(msg: SetState to 2)(5962 ms)]           
2019-06-16 20:23:17 +0200 Assertions          	PID 53(powerd) Created InternalPreventSleep "com.apple.powermanagement.acwakelinger" 00:00:00  id:0x0xd00008ea8 [System: DeclUser BGTask SRPrevSleep kCPU kDisp]          
Sleep/Wakes since boot at 2019-06-16 16:42:21 +0200 :5   Dark Wake Count in this sleep cycle:1

Time stamp                Domain              	Message                                                                    	Duration  	Delay     
==========                ======              	=======                                                                    	========  	=====     
UUID: 6A6F16C8-CEAF-4945-A7AC-DF63FDD49242
2019-06-16 20:23:21 +0200 Notification        	Display is turned off                                                      	          
2019-06-16 20:23:21 +0200 Assertions          	PID 1075(rcd) Summary UserIsActive "com.apple.rcdevent" 01:59:25  id:0x0x900008d00 [System: DeclUser BGTask SRPrevSleep IPushSrvc kCPU kDisp]          
2019-06-16 20:23:21 +0200 Assertions          	PID 1075(rcd) Summary UserIsActive "com.apple.rcdevent" 01:59:16  id:0x0x900008d0d [System: DeclUser BGTask SRPrevSleep IPushSrvc kCPU kDisp]          
2019-06-16 20:23:21 +0200 Assertions          	PID 93(hidd) Summary UserIsActive "com.apple.iohideventsystem.queue.tickle.4294968377.3" 00:00:00  id:0x0x9000080f0 [System: DeclUser BGTask SRPrevSleep IPushSrvc kCPU kDisp]          
[... cut ...]
2019-06-16 20:24:20 +0200 Assertions          	PID 53(powerd) Created InternalPreventSleep "com.apple.powermanagement.acwakelinger" 00:00:00  id:0x0xd00008edc [System: DeclUser BGTask SRPrevSleep kCPU kDisp]               
`

Showing `Wake from Normal Sleep [CDNVA] due to PEG1 PEGP PEG2 PEGP PXSX PXSX/UserActivity Assert: Using AC (Charge:0%) 4 secs` at 20:23:17 (this is using WOL).

Then put to sleep again, and immediatly waking up:
`Wake from Normal Sleep [CDNVA] due to /: Using AC (Charge:0%)` at 20:24:20 (immediatly waking up)

Mieze, 2019-06-16:
I think this background task prevents the machine from falling asleep after a darkwake:

`   pid 312(cloudphotosd): [0x00001bef000b8ec0] 00:01:32 BackgroundTask named: "com.apple.cloudphotosd.darkwake.sync" 
	Details: cloudphotosd darkwake power assertion for initial and large iCPL downloads/uploads
	Timeout will fire in 1708 secs Action=TimeoutActionTurnOff
`

jockebq, 2019-06-16:
@Mieze So I disabled iCloud Photos in the settings of the Photos app.
Same thing happens, it wakes up immediatly.
Now it is not showing in the log the pid 12(cloudphotosd) in the log. But the issue persists:

```
Time stamp                Domain              	Message                                                                    	Duration  	Delay     
==========                ======              	=======                                                                    	========  	=====     
UUID: 6C4DFD92-0268-4268-8973-594600253D18
2019-06-16 20:51:05 +0200 Notification        	Display is turned off                                                      	          
2019-06-16 20:51:05 +0200 Assertions          	PID 421(AddressBookSourceSync) Summary PreventUserIdleSystemSleep "Address Book Source Sync" 00:00:07  id:0x0x100008180 [System: PrevIdle DeclUser BGTask IPushSrvc kDisp]          
2019-06-16 20:51:05 +0200 Assertions          	PID 93(hidd) Summary UserIsActive "com.apple.iohideventsystem.queue.tickle.4294968387.3" 00:00:00  id:0x0x900008182 [System: PrevIdle DeclUser BGTask IPushSrvc kDisp]          
2019-06-16 20:51:05 +0200 Assertions          	PID 73(apsd) Summary ApplePushServiceTask "com.apple.apsd-recreatecourierstate-push.apple.com" 00:00:18  id:0x0xb00008083 [System: PrevIdle DeclUser BGTask IPushSrvc kDisp]          
2019-06-16 20:51:05 +0200 Assertions          	PID 63(mds) Summary BackgroundTask "com.apple.metadata.mds.power" 00:00:21  id:0x0xb0000801a [System: PrevIdle DeclUser BGTask IPushSrvc kDisp]          
2019-06-16 20:51:05 +0200 Assertions          	PID 73(apsd) Summary InteractivePushServiceTask "com.apple.apsd-outgoingmessage" 00:00:10  id:0x0x120000811a [System: PrevIdle DeclUser BGTask IPushSrvc kDisp]          
2019-06-16 20:51:05 +0200 Assertions          	PID 73(apsd) Summary InteractivePushServiceTask "com.apple.apsd-recreatecourierstate-sandbox.push.apple.com" 00:00:09  id:0x0x120000814c [System: PrevIdle DeclUser BGTask IPushSrvc kDisp]          
2019-06-16 20:51:05 +0200 Assertions          	Kernel Idle sleep preventers: -None-          
2019-06-16 20:51:05 +0200 Assertions          	PID 53(powerd) Created InternalPreventSleep "com.apple.powermanagement.darkwakelinger" 00:00:00  id:0x0xd00008184 [System: PrevIdle DeclUser BGTask SRPrevSleep IPushSrvc kCPU kDisp]          
2019-06-16 20:51:06 +0200 Assertions          	PID 73(apsd) Released InteractivePushServiceTask "com.apple.apsd-recreatecourierstate-sandbox.push.apple.com" 00:00:10  id:0x0x120000814c [System: PrevIdle DeclUser BGTask SRPrevSleep IPushSrvc kCPU kDisp]          
2019-06-16 20:51:07 +0200 Assertions          	PID 73(apsd) Released ApplePushServiceTask "com.apple.apsd-recreatecourierstate-push.apple.com" 00:00:20  id:0x0xb00008083 [System: PrevIdle DeclUser BGTask SRPrevSleep IPushSrvc kCPU kDisp]          
2019-06-16 20:51:07 +0200 Assertions          	PID 73(apsd) Released InteractivePushServiceTask "com.apple.apsd-outgoingmessage" 00:00:13  id:0x0x120000811a [System: PrevIdle DeclUser BGTask SRPrevSleep IPushSrvc kCPU kDisp]          
2019-06-16 20:51:12 +0200 Assertions          	PID 227(diskmanagementd) Released PreventUserIdleSystemSleep "com.apple.diskmanagementd" 00:00:26  id:0x0x10000801f [System: PrevIdle DeclUser BGTask SRPrevSleep kCPU kDisp]          
2019-06-16 20:51:20 +0200 Assertions          	PID 53(powerd) TimedOut InternalPreventSleep "com.apple.powermanagement.darkwakelinger" 00:00:15  id:0x0xd00008184 [System: PrevIdle DeclUser BGTask SRPrevSleep kCPU kDisp]          
2019-06-16 20:51:20 +0200 Assertions          	Summary- [System: PrevIdle DeclUser BGTask kDisp] Using AC          
2019-06-16 20:51:20 +0200 Sleep               	Entering Sleep state due to 'Software Sleep pid=87': Using AC (Charge:0%) 22 secs   
2019-06-16 20:51:22 +0200 Wake Requests       	[*proc=mDNSResponder request=Maintenance inDelta=7197]           
2019-06-16 20:51:22 +0200 PM Client Acks      	Delays to Sleep notifications: [com.apple.apsd is slow(2524 ms)]           
2019-06-16 20:51:36 +0200 Kernel Client Acks  	Delays to Sleep notifications: [AppleHDADriver driver is slow(msg: SetState to 0)(1013 ms)] [AMDFramebufferCI driver is slow(msg: SetState to 1)(2098 ms)] [AMDFramebufferCI driver is slow(msg: SetState to 1)(2145 ms)] [en0 driver is slow(msg: WillChangeState to 0)(2355 ms)]           
2019-06-16 20:51:36 +0200 Assertions          	Kernel Idle sleep preventers: IODisplayWrangler          
2019-06-16 20:51:36 +0200 Assertions          	PID 212(mDNSResponder) Created MaintenanceWake "mDNSResponder:maintenance" 00:00:00  id:0x0xd00008196 [System: PrevIdle DeclUser BGTask SRPrevSleep kCPU kDisp]          
2019-06-16 20:51:36 +0200 Notification        	Display is turned on                                                       	          
2019-06-16 20:51:36 +0200 Assertions          	PID 212(mDNSResponder) Released MaintenanceWake "mDNSResponder:maintenance" 00:00:00  id:0x0xd00008196 [System: PrevIdle DeclUser BGTask kDisp]          
2019-06-16 20:51:41 +0200 Notification        	Next immediate inactivity window start:'2019-06-16 21:15:00 +0200' end:'2019-06-16 21:45:00 +0200'	          
2019-06-16 20:51:42 +0200 Wake                	Wake from Normal Sleep [CDNVA] due to PEG1 PEGP PEG2 PEGP PXSX PXSX/Network: Using AC (Charge:0%) 4 secs    
2019-06-16 20:51:42 +0200 HibernateStats      	hibmode=0 standbydelay=86400                                               	          114       	
[... cut ...]
   id=508  level=255 0x4=USB mod=1970-01-01 01:00 description=com.apple.usb.externaldevice.1d180000 owner=General Purpose USB Hub
   id=510  level=255 0x4=USB mod=1970-01-01 01:00 description=com.apple.usb.externaldevice.1d182000 owner=SteelSeries H Wireless
   id=511  level=255 0x4=USB mod=1970-01-01 01:00 description=com.apple.usb.externaldevice.1d183000 owner=SteelSeries H Wireless
   id=512  level=255 0x100=MAGICWAKE mod=2019-06-16 21:08 description=en0 owner=en0
Idle sleep preventers: IODisplayWrangler

```

Any other suggestions? Power nap is disabled in Settings, and I use darkwake=0.

jockebq, 2019-06-16:
I noticed a difference between the two sleeps.
The sleep that works (the first one) logs this:
`Entering Sleep state due to 'Software Sleep pid=87'`

The second sleep (after WOL)
`Entering DarkWake state due to 'Software Sleep pid=87'`

So there seems to be a different sleep it tries after using WOL. Is this information useful to solve this?

Mieze, 2019-06-16:
Maybe yes but I must admit that I'm currently at a loss.

jockebq, 2019-06-16:
Me too, but I have tried everything I can think of, changed all the Power saving settings. Tried darkwake=0, 1, 2
(right now I use darkwake=0).

Seems like the WOL triggers some kind of change in the sleep though, as it goes to darkwake instead of sleep?

jockebq, 2019-06-26:
I have done as much testing as I can think of. I even removed all the _PRW from my DSDT.aml. It still wakes up from sleep even if all _PRW is removed.
I also tried all the versions of this kext with the same result.

My only conclusion is that using Wake On LAN triggers darkwake to be enabled again, as it will try to go into DarkWake state when sleeping after using WoL. I have both the bootflag `darkwake=0` and `sudo pmset -a darkwakes 0`. So is it something in this kext that might trigger darkwake to be reset?

jockebq, 2019-06-26:
Maybe I found the issue, from the original drivers.
Check here:
https://lore.kernel.org/patchwork/patch/904931/
And here:
https://bugzilla.kernel.org/show_bug.cgi?id=61651#c126

Do you think updating the source code would help?

Mieze, 2019-06-28:
No, this doesn't have anything to do with your problem because this is an issue which I had solved almost 5 years ago.
Once again, the driver programs the NIC to cause a wakeup event on reception of a Magic Packet but your wakeup isn't caused by the NIC, which means that you are on a completely wrong track. The driver is not the cause of the problem!!! You are wasting your time. Search somewhere else!

closed 2019-06-28 by Mieze

marcin-fr, 2020-12-07:
Did you find the solution to this problem? I have exactly the same symptoms.

> Hi,
>
> I have Wake On LAN from sleep working great. Except one issue.
> After I have woken the hackintosh with wake on lan, and put it to sleep again, it will immidiatly wake up. This never happens when [...]

> Hi,
>
> I have Wake On LAN from sleep working great. Except one issue.
> After I have woken the hackintosh with wake on lan, and put it to sleep again, it will immidiatly wake up. This never happens when [...]