Mieze/AtherosE2200Ethernet #21: Jumbo 9K not work

Brax84, 2019-10-24:
@Mieze First of All big thanks for your work!

I've a problem with Jumbo frames enabled. My pc not open network folder when setting Jumbo Frames. Under the  network i can see my Nas Synology but can't open folders on Nas. I get a network error message. With MTU 1500 no problem.
when I start the hack with jumbo frames enabled, I can browse on google, but
if I try to mount a network folder on the NAS, Google or other sites no longer open.
On windows Jumbo 9K work very well can get 65-70Mb/s without Jumbo enable can get 30-35Mb/s.
On windows works well with this setting [LINK](https://support.killernetworking.com/knowledge-base/optimizing-e2200-e2400-e2500-advanced-driver-settings/)

I've put your kext in Clover /kext/other
Set boot with npci=0x2000 or npci=0x3000 
Test all Duplex but without luck 

Attach my kernel logs and netstat -s 

[log.zip](https://github.com/Mieze/AtherosE2200Ethernet/files/3767468/log.zip)

Thanks in advance

p.s sorry for my bad English

Mieze, 2019-10-24:
> @Mieze First of All big thanks for your work!
> 
> On windows works well with this setting [LINK]

If you followed the instructions in the link above, you've got a completely misconfigured network because these instructions are nonsense and the author has no absolutely no idea what he or she is talking about.

By the way, there is no need to use jumbo frames at all and the driver is able to handle speeds of more than 100 MB/s with default settings. Fix your network, and most likely your NAS too, and everything will be fine.

Brax84, 2019-10-24:
Thanks for suggestions. I will investigate to fix my network
Thanks

closed 2019-10-24 by Brax84