Mieze/AtherosE2200Ethernet #22: Killer Ethernet E2500v2 - DELL G5-5590(2019)

radaelilucca, 2019-12-24:
Is it the kext for this adapter? There is a lot of people saying me to use AtherosE2200Ethernet, but it's not working for me... I try a lot of times, but i can't get it work...

Mieze, 2019-12-29:
Your problem description doesn't sound like a driver issue, looks more like a user error. In case you need help please use the driver's thread on insanelymac.com: https://www.insanelymac.com/forum/topic/300056-solution-for-qualcomm-atheros-ar816x-ar817x-and-killer-e220x/

closed 2019-12-29 by Mieze

drewtyliuk, 2020-10-26:
> Is it the kext for this adapter? There is a lot of people saying me to use AtherosE2200Ethernet, but it's not working for me... I try a lot of times in Catalina, but i can't get it work...

I have the same system and same problem. Were you able to find a solution?

radaelilucca, 2020-10-26:
> > Is it the kext for this adapter? There is a lot of people saying me to use AtherosE2200Ethernet, but it's not working for me... I try a lot of times in Catalina, but i can't get it work...
> 
> I have the same system and same problem. Were you able to find a solution?

Hey yo!

For this model you have to use the [RealtekRTL8111.kext](https://github.com/Mieze/RTL8111_driver_for_OS_X). It works 100%!

If you are newbie like me, I know a guy who is very good at it and keeps his EFI updated and documented here on github. Repo: [Hackintosh-Dell-G5-5590](https://github.com/leocg/Hackintosh-Dell-G5-5590)

His model is the GTX 2060 one (and mine is GTX 1660ti) but I'm using this EFI without any problems.

drewtyliuk, 2020-10-27:
> Hey yo!
>
> For this model you have to use the [RealtekRTL8111.kext](https://github.com/Mieze/RTL8111_driver_for_OS_X). It works 100%!
>
> If you are newbie like me, I know a guy who is very good at it [...]

So I actually read the number wrong and I have the desktop 5090 but the the still kext worked 100% so thank you so much. I would have been looking forever.