Mieze/AtherosE2200Ethernet #23: Keep v2.1.0 available in releases

heisian, 2019-12-29:
As documented in [this thread](https://www.tonymacx86.com/threads/ethernet-not-working-high-sierra-z97-g45-gaming-killer-e2205-gigabit.268860/), the latest release, v2.2.2, does _not_ work with the MSI Z97 G45's E2205 chip. Installing the v2.1.0 release works great. It would be nice if the old release was available for folks like me.

Thank you!!!

Mieze, 2019-12-29:
The E2005 is working fine with version 2.2.2. I verified it myself on a friends machine. In case you've got problems, there must be something wrong with your system configuration (wrong driver version for OS, etc.).

closed 2019-12-29 by Mieze

heisian, 2019-12-30:
Configurations can vary pretty widely from system to system... I'm not the only one with the issue, just look at the thread. With the _exact same_ EFI configuration and kexts, downgrading to v2.1.0 makes ethernet work on my board. https://github.com/heisian/Z97-G45-GAMING-macOS

Mieze, 2019-12-30:
Actually the thread you mentioned contains virtually nothing which indicates a driver bug but looks more like a bunch of users which aren't smart enough to make it work. Sorry, but that's the way it is!

heisian, 2019-12-31:
Alright, well, now you're insulting people's intelligence. Clearly there is nothing productive left to be said here. Have a great day.