Mieze/AtherosE2200Ethernet #25: [Bug/Regression] v2.1.0 - v2.2.2

heisian, 2020-01-05:
Using this [EFI configuration](https://github.com/heisian/Z97-G45-GAMING-macOS), and without making any changes between boots, the following result occurs:

AtherosE2200Ethernet v2.2.2 - **Does not work**:
[image]

AtherosE2200Ethernet v2.1.0 - **Works**:
[image]

Mieze, 2020-01-05:
What are you trying to tell me? And, most important of all, why haven't you checked the kernel logs as this is the point to start with in case there is a problem?

heisian, 2020-01-05:
I think the title is pretty clear, as is the configuration and the conditions under which the latest kext version does not work. I'll see if there's anything useful in the kernel logs.