Mieze/AtherosE2200Ethernet #26: VLANs don't appear to be working

inickt, 2020-05-29:
Hi @Mieze! Thanks for your work on this driver. I have been using it for the past 4 or so years with basically no problems.

I recently have been using VLANs a bunch with my homelab and have had some issues with this kext. I have added a virtual VLAN interface in network settings and have tagged it on 99. However, I am unable to reach any devices on that VLAN. I can see traffic go out in Wireshark when I ping a device, but I get no response. Right now I have an external USB adapter that I can use, but I would love to get my internal NIC working so I don't have to swap ethernet cords/get a switch just for my management network. If it matters I am running a GIGABYTE Z97MX-Gaming 5 which has the Killer E2200 NIC in it (and I am using the latest v2.3.0 release).

I know it says in the README that VLANs are untested, but if there is anything I can do to help debug please let me know. I ran `grep kernel /var/log/system.log` after switching to a debug version but I actually don't see any logs. I briefly looked through the upstream linux driver and nothing is jumping out to me. That being said I would love to work on getting a fix for this, and I'll try to play around with building the kext and contributing an upstream patch if I am able to fix it. With that being said I have never done any sort of kext development and my C/C++ is a little rusty- do you mind pointing me in the right direction?
1. Is there an easy way of running the kext without having to build and replace my version in my OpenCore config?
2. Off the top of your head are there any specific places you can think that might be worth looking at/debugging to fix VLANs? 

Thanks for your help!

Mieze, 2020-06-06:
I'm going to take a look at the problem this weekend.

Mieze, 2020-06-08:
Try version 2.3.2 which fixes the issue.

inickt, 2020-06-09:
Thank you for taking a look! I build and exported a kext using Xcode (couldn't find a prebuilt download for 2.3.2 yet) but I am actually seeing the same issue. I can see ICMP packets leaving my machine with Wireshark on the VLAN interface, but my interface is not receiving any responses.

Tried both debug and release variants, and I don't see any logs coming up in `/var/log/system.log`. Is there anything I can do to help debug? Thanks!

Mieze, 2020-06-09:
First of all, please see the help information in the driver's thread in the insanelymac.com forum, in particular in order to learn how to retrieve kernel logs. Second, keep in mind that all VLAN traffic is going through a virtual interface called _vlan0_ or something like that. I personally verified that it works, with a Macbook pro and my Netgear R7800 running OpenWRT.

Also make sure that your system is using the new version with kextstat in Terminal.