Mieze/AtherosE2200Ethernet #28: Kext not working for Killer E2500

fmartis00, 2020-11-03:
Hi, as the title says kext is not working, and as You can see from the attached log it dont' even appears to be loaded.
I tried different versions, from 2.2 up to the last one.
If You need me to do some tests, feel free to ask.

Thank You in advance.

My hardware: DELL G3 3500
i7 10750h
16gb ram ddr4
Killer E2500
Ax1650i Wifi 6

[New Text Document.txt](https://github.com/Mieze/AtherosE2200Ethernet/files/5484437/New.Text.Document.txt)

Mieze, 2020-11-03:
Try RealtekRTL8111.kext because I'm not sure if the NIC that DELL calls Killer E2500V2 is really an Atheros chip. At least some users reported similar issues: Please see: [https://github.com/Mieze/AtherosE2200Ethernet/issues/22](url)

Emanuele-1998, 2020-11-03:
> Try RealtekRTL8111.kext because I'm not sure if the NIC that DELL calls Killer E2500V2 is really an Atheros chip. At least some users reported similar issues: Please see: [https://github.com/Mieze/AtherosE2200Ethernet/issues/22](url)

Work with this https://www.insanelymac.com/forum/applications/core/interface/file/attachment.php?id=331774

Thanks for the job 🙏

fmartis00, 2020-11-03:
Thanks for the fast replies. I solved using Realtek kext! I Appreciate Your effort:)

closed 2020-11-03 by fmartis00