Mieze/AtherosE2200Ethernet #29: Kernel Panis using version 2.3.3

RoninSpectre, 2020-12-04:
Updated to Open Core 6.3 and also updated the kext to version 2.3.3 which gave me a kernel panic. Rolled back to 2.3.2 which solved my problem. Running a Fatal1ty Z170 Gaming K4, i7 6700K. 
[image]

Mieze, 2020-12-04:
Clean system caches to resolve the problem as it's a linker error.

closed 2020-12-04 by Mieze

RoninSpectre, 2020-12-04:
Cleaned system cache and the same error persists.
https://i.postimg.cc/pL71rjq3/EA6-DC86-F-BAF8-4656-8582-4007-AA7-EE6-F6-1-105-c.jpg

Mieze, 2020-12-04:
I don't know what's wrong with your system but kernel panic is definitely unrelated to the driver because it indicates an ACPI problem. That's the reason why I can rule out a driver issue.