Mieze/AtherosE2200Ethernet #30: After last merge AtherosE2200Ethernet won't work if built when downloading your repository.

tecnicasopr, 2020-12-09:
I'm booting fine with your latest AtherosE2200Ethernet 2.3.3 release. The problem is it only boots with the kext you built and uploaded here. But my system won't boot when downloading your repository and building it with xcode, or with the help of building tools within OC builder or Hackintool.

Here is a pic of the screen. Usually I try to take screenshots within OpenCore but this time I couldn't.
[image]
It also fails when building with ocbuilder or Hackintool. I suspect the problem started when you merged. 
Thanks in advance for your work, 
                                                     Sincerely,
                                                                    Tecnicasopr

Mieze, 2020-12-09:
Looks like you are having a build problem?

By the way, what makes you assume that it has anything to do with the driver because in your screenshot there is no reference to AtherosE2200Ethernet?

tecnicasopr, 2020-12-09:
You are right, the screenshot doesn't shows any reference to your kext. But
I'm sure it's your repo because when I download the latest version you
built and upload on insanlymac  it works. Then when building my self with
code or with the help of ocbuilder or hackintool it fails.

closed 2021-06-28 by Mieze