Mieze/AtherosE2200Ethernet #34: disable eee

seppelinchen, 2022-05-12:
hi,
i have an issue with short network outages. 3-6 packets get lost from time to time. you sugest, disabling eee to solve that.
but how do i disable it?
thank you...

ganthore, 2022-05-12:
I'm not on my main hack right now but I believe you're supposed to see a drop down option for it in your network adapter settings under the advanced button, then navigate to the hardware tab. Set your config to manual and look at the duplex drop down. One option will be set to something like full duplex, energy efficient. Just change it to the a similar entry that lacks the energy efficient option.

seppelinchen, 2022-05-17:
found it.
but unfortunately it doesn't solve the problem. still from time to time some packets get lost and teamviewer disconnects :(

closed 2025-02-08 by Mieze